byuccl/RapidSmith2 #169: BelRouteThrough -- verify input and output are on the same BEL

trharoldsen, 2017-01-07:
The BELRouteThrough is a class created by users to indicate that a BEL acts as a routethrough.  The constructor takes two pins, an input and an output pin but never validates the pins are on the same BEL.  In the constructor, we could add a check to guarantee the pins are on the same BEL.

This would also allow us to obtain the BEL straight from the pins without requiring the Bel be passed as an additional argument to the constructor.  We could allow then remove the Bel field and just obtain the Bel through either the input or the output pin (we'd've guaranteed they're on the same Bel at this point).

ttown523, 2017-01-09:
This is a good idea.

trharoldsen, 2017-01-11:
Resolved in #173

closed 2017-01-11 by trharoldsen