byuccl/RapidSmith2 #171: Do we want to replicate RapidSmith's DesignExplorer program?

nelsobe, 2017-01-09:
DesignExplorer works on XDL designs so we have removed it in pull request #164.

1. Do we want to create a new version of it for RS2?  
2. What would be the value of it?  
3. Would we want to use the same QT package for graphics or is it time to move to something newer?
4.  Would we want it to have identical behavior to the original, or would we want it to have different behavior in some ways?

ttown523, 2017-01-09:
If we do this, I think it would be good to migrate to a 64-bit QT package for windows. It can be a tad annoying having to switch to a 32-bit JRE just to run the GUI stuff on windows.

DallonTG, 2018-05-16:
See #262 on 64-bit QT.