byuccl/RapidSmith2 #191: Updates to RouteTree class

trharoldsen, 2017-01-17:
setSourceTree should be private -- it can be accomplished with the addConnection(Connection, RouteTree) method.
setConnection should be private -- same as above
Remove cost field from tree -- This method is only seldom used, takes up space and an int is not always what the user wants (the may want double).  For cases where cost is useful, they can create a map of trees -> costs or extend RouteTree
Make RouteTree extensible -- extending the class would require reimplementing the addConnection class to create an instance of the RouteTree subclass.  We could probably due something with creating an abstract base class which the user can extend that doesn't implement the addConnection method forcing them to extend it.

mcrossen, 2017-04-09:
Making the RouteTree class extensible will basically require changing the entire class. The difficulty comes from all the methods where RouteTree references itself in both return and parameter types. Sadly, Java has poor support for recursive inheritable data structures. We'll talk more the next time we meet. I have some ideas we can talk about then.

trharoldsen, 2017-05-10:
see #231

trharoldsen, 2017-12-25:
Reference pr #323

DallonTG, 2018-05-01:
This seems resolved so I'm closing it.

closed 2018-05-01 by DallonTG