byuccl/RapidSmith2 #208: Adding BelType enum to Bel objects

ttown523, 2017-02-24:
Currently in RapidSmith, there is no `BelType` enumeration to distinguish BELs of a certain type. The current workaround for this is doing a string comparison with the BEL name for what you are looking for ("LUT" for example). Vivado has a property called TYPE associated with each BEL object, that can be used for a `BelType` in RS2, this info needs to be added to the FamilyInfo.xml file.

nelsobe, 2017-02-24:
I agree - this would be a good addition.  

What would be a good way of prioritizing issues?  This would be one that would seem to be high priority (and also easy to accomplish by one of our new student researchers).

trharoldsen, 2017-05-07:
#232

DallonTG, 2018-05-03:
BelTypes are included in the familyInfo.xml file, but there is still no BelType enumertion. I think this would still be a good addition.