byuccl/RapidSmith2 #212: Changing SiteType causes issues with parallel code

trharoldsen, 2017-03-11:
Right now, the site stores its last configured type in a template property.  While this likely improves performance as it avoids looking up the template every time we need it, it causes issues with parallel code.  If multiple designs are loaded, the designs will fight over the site type of a site and will execute with the expectation that the site is configured to the same type.

This is exacerbated by us caching a device and the designs by default (and without an easy way to change the operation) load the same device object.

As far as I can recall, this siteTemplate object is the only mutable property in the device.  I propose that we cease relying on the siteTemplate property and make all site-level queries use the getObject(name, SiteType) form which will guarantee that the designs are never fighting over the site type of the object.

ttown523, 2017-03-13:
Design Export Updates:

- The function `site.getType()` is used to set the site type in Vivado. However, there is no guarantee that the design data structures match the device data structures. We need to use the function call `cell.getBel().getId().getType()` instead.

- Add a check to make sure that types are not being mixed for a single site. An error should be thrown if this happens.

Design Import Updates:

- Replace all function calls `site.getBel(belName)` with site.getBel(belName, siteType). This will ensure that we are getting the write Bel object when applying placement in RS2.

trharoldsen, 2017-05-10:
Support added in #241

closed 2017-05-10 by trharoldsen