byuccl/RapidSmith2 #224: Control sets

trharoldsen, 2017-04-24:
Control sets are properties of a site in which all cells in a site must share the same property.  We should create a way of defining the control sets and exposing which properties on the cells relate to the control sets.

trharoldsen, 2017-04-28:
I'm going to need this to port the packer over.

ttown523, 2017-06-08:
I don't think there is a good way to detect control sets in Vivado. There is only one Vivado command "report_control_sets" that reports the available control sets. We may be able to parse this string, but there is no way to automatically determine mappings from cell properties to site properties.

```
Copyright 1986-2017 Xilinx, Inc. All Rights Reserved.
------------------------------------------------------------------------------------
| Tool Version : Vivado v.2017.1 (win64) Build 1846317 Fri Apr 14 18:55:03 MDT 2017
| Date         : Thu Jun  8 08:52:14 2017
| Host         : CB461-EE09968 running 64-bit Service Pack 1  (build 7601)
| Command      : report_control_sets -verbose
| Design       : VGA
| Device       : xc7a100t
------------------------------------------------------------------------------------

Control Set Information

Table of Contents
-----------------
1. Summary
2. Flip-Flop Distribution
3. Detailed Control Set Information

1. Summary
----------

+----------------------------------------------------------+-------+
|                          Status                          | Count |
+----------------------------------------------------------+-------+
| Number of unique control sets                            |    35 |
| Unused register locations in slices containing registers |    72 |
+----------------------------------------------------------+-------+

2. Flip-Flop Distribution
-------------------------

+--------------+-----------------------+------------------------+-----------------+--------------+
| Clock Enable | Synchronous Set/Reset | Asynchronous Set/Reset | Total Registers | Total Slices |
+--------------+-----------------------+------------------------+-----------------+--------------+
| No           | No                    | No                     |             160 |           41 |
| No           | No                    | Yes                    |              13 |            8 |
| No           | Yes                   | No                     |              28 |           12 |
| Yes          | No                    | No                     |             150 |           44 |
| Yes          | No                    | Yes                    |             225 |           77 |
| Yes          | Yes                   | No                     |              32 |           12 |
+--------------+-----------------------+------------------------+-----------------+--------------+

3. Detailed Control Set Information
-----------------------------------

+--------------+--------------------------------+------------------------------------------------+------------------+----------------+
| Clock Signal |          Enable Signal         |                Set/Reset Signal                | Slice Load Count | Bel Load Count |
+--------------+--------------------------------+------------------------------------------------+------------------+----------------+
|  PCLK        |                                | FSM_sequential_statemachine.c_state[2]_i_1_n_0 |                2 |              2 |
|  CLK_I       | rptr_rep[3]_i_1_n_0            |                                                |                1 |              4 |
|  CLK_I       | rptr_rep[3]_i_1_n_0            | wptr[3]_i_1__0_n_0                             |                1 |              4 |
|  CLK_I       |                                | WB_block.CYC_O_i_2_n_0                         |                2 |              4 |
|  CLK_I       | wptr[3]_i_2_n_0                | wptr[3]_i_1__0_n_0                             |                1 |              4 |
|  PCLK        | u3/tblk.vtgen/ver_gen/p_0_in   |                                                |                4 |              4 |
|  CLK_I       |                                | ctrl[15]_i_2_n_0                               |                3 |              5 |
|  CLK_I       |                                | pixel_buf/fifo_cnt[4]_i_1_n_0                  |                2 |              5 |
|  CLK_I       | hvlen[31]_i_1_n_0              | ctrl[15]_i_2_n_0                               |                1 |              5 |
|  CLK_I       | line_fifo_wreq                 | FSM_sequential_statemachine.c_state[2]_i_1_n_0 |                3 |              8 |
|  PCLK        |                                | u3/nVen                                        |                3 |              8 |
|  PCLK        | Qi[7]_i_1__5_n_0               |                                                |                2 |              8 |
|  PCLK        | cgate                          | FSM_sequential_statemachine.c_state[2]_i_1_n_0 |                2 |              8 |
|  CLK_I       | ctrl[31]_i_1_n_0               | ctrl[15]_i_2_n_0                               |                6 |             15 |
|  CLK_I       | WB_block.hgate_cnt[15]_i_1_n_0 |                                                |                5 |             16 |
|  CLK_I       | WB_block.vgate_cnt[15]_i_1_n_0 |                                                |                5 |             16 |
|  CLK_I       | htim[31]_i_1_n_0               | WB_block.CYC_O_i_2_n_0                         |                5 |             16 |
|  CLK_I       | htim[31]_i_1_n_0               | ctrl[15]_i_2_n_0                               |                7 |             16 |
|  CLK_I       | vtim[31]_i_1_n_0               | WB_block.CYC_O_i_2_n_0                         |                5 |             16 |
[... cut ...]
|  CLK_I       |                                |                                                |               25 |            102 |
+--------------+--------------------------------+------------------------------------------------+------------------+----------------+

+--------+-----------------------+
| Fanout | Number of ControlSets |
+--------+-----------------------+
| 2      |                     1 |
| 4      |                     5 |
| 5      |                     3 |
| 8      |                     4 |
| 15     |                     1 |
| 16+    |                    21 |
+--------+-----------------------+ 
```

DallonTG, 2018-05-23:
@trharoldsen, is there still a need for this or would it provide much benefit for your packer still?

trharoldsen, 2018-05-23:
It would be useful, but in terms of the packer, I think the current
approach is to just add an architecture dependent rule that knows what they
are and ensures we're not violating any control sets.