byuccl/RapidSmith2 #246: Convert Wires To Nodes

trharoldsen, 2017-05-12:
As of now, RS2 treats the same wire in different tiles as different wires.  This works fine for routing but likely slows down the router as it must look at two wires for each PIP it traverses.  We could group all of the wires that are part of the same metal into a single node, reducing the size of the route trees that are represent the wires.

GTRI-nathan, 2017-11-29:
I believe a problem that I'm running into is that the same node (but as different wires) is being used twice when routing a design, which of course gives errors in Vivado. Is there a way to determine if two wires are from the same node currently? I might be misunderstanding, but it seems like this accidental double use of a wire could crop up a lot in crowded designs.

trharoldsen, 2017-11-30:
@nelsobe @mjc31511 I know some people have dealt with this while creating routers.  Do you have any suggestions, Michael? or does Dr. Wirthlin?

GTRI-nathan, 2017-11-30:
I've worked around it for now by always checking against the closure of wires that can be reached by non PIP connections (but allowing them if they haven't been used before and the previous wire is part of the same segment), but thought I should bring it up in case I was missing the correct way to do it / comment that representing routing with nodes instead of wires would nice from my perspective.

mjc31511, 2017-11-30:
@GTRI-nathan I do something similar to ensure that two routes do not use the same node. This is mostly (if not entirely) an issue with long lines since they can be driven by multiple sources. Other than checking the closure of wires that can be reached by non PIP connections or forcing a route to include all wires that can be reached by these connections, I see no other way around the issue. I agree that it would be a major improvement to use a node representation instead of the current wire representation.

GTRI-nathan, 2017-11-30:
Thanks for clarifying!