byuccl/RapidSmith2 #247: sourceSitePin in CellNet never updates

mjc31511, 2017-05-26:
The sourceSitePin never updates when unplacing and unrouting the design. I found this while changing the placement of a few cells and trying to route them, this was returning the old site pin. I think it would be logical for it to change at some point. Right now I have written a workaround (I'm not saying this is the best way to do it). When mapping the cell pin to a new bel pin, I check if it is the source pin of a net and update the sourceSitePin of the net if it is (see following code found in CellPin.java):

```
public boolean mapToBelPin(BelPin pin) {
		
		if (cell.isMacro()) {
			throw new Exceptions.DesignAssemblyException("Cannot map external macro pins to bel pins. Only leaf pins can be mapped to bel pins"); 
		}
		
		if (belPinMappingSet == null) {
			belPinMappingSet = new HashSet<>();
		}
		
		//User added code
		if(this.getNet().getAllSourcePins().size() == 1){ //if this is the only source pin
			if(this.getNet().getSourcePin().equals(this)){ //if this is the net's source pin
				//remap the the SitePin
				if(this.getNet().getSourceSitePin() != null){ //if there existed source site pin
					if(!this.getCell().getSite().equals(this.getNet().getSourceSitePin().getSite())){ //and the site has changed (e.g. this cell is on a new site)
						this.getNet().setSourceSitePin(this.getCell().getSite().getSitePin(this.getNet().getSourceSitePin().getName())); //set to "same" site pin on the new site
					}
				}
			}
		}
		
		return belPinMappingSet.add(pin);
	}
```

I think this workaround should work given that the sites retain the same packing and remain on the same type of site (this is why this may not be the best place to put it). At the very least, when the net is unrouted the sourceSitePin can be set to null.

trharoldsen, 2017-11-06:
This may have been fixed.  Can someone check up on this.

DallonTG, 2017-11-22:
Addressed in PR #308.

closed 2017-11-22 by DallonTG