byuccl/RapidSmith2 #248: Bonded IOB Sites Don't Exist (xc7a100t-csg324)

iamthehorker, 2017-05-31:
Only unbonded (which you can't do much with) IOB sites are returned with the following:
```
for (Object key: device.getSites().keySet())
    System.out.println(key)
```
For example, the following are some Sites that do not exist/return a value when called by device.getSite(). They are also not unbonded, and every other IOB Site that is not unbonded also does not return a value.
IOB_X0Y149
IOB_X0Y183

The following unbonded IOB does return a value. Only unbonded sites seem to exist in your device files.
IOB_X0Y49

Perhaps there was a mistake, and only unbonded sites were included instead of the other way around?
[image]

ttown523, 2017-05-31:
For series7 devices, the package pin name (G13) for bonded sites is used instead of the IOB site name (X0Y149). Unbonded sites use the regular site name since there is no corresponding package pin. 

The function call `device.getSite("G13")` will return a handle to the site referenced in the figure above.

The reason for this naming is that ISE XDLRC files used this naming convention, so that is how we represent it in RapidSmith as well for series7 devices.

iamthehorker, 2017-05-31:
Ahh, I see. I wasn't unaware of that. I assumed bonded and unbonded IOB sites used the same naming scheme.

closed 2017-05-31 by iamthehorker