byuccl/RapidSmith2 #252: Vivado Interface does not import correct route

mjc31511, 2017-07-12:
It looks like the Routing import function, found in XdcRoutingInterface.java, recreateRoutingNetwork(CellNet, String, Set<String>) does not import the route correctly from Vivado. As seen in the routing.xdc file, Vivado specifies one route, but when I walk through the route tree in RapidSmith, I get a different route.

Route from Vivado
> { CLBLM_L_AQ CLBLM_LOGIC_OUTS0  { SR1BEG1 IMUX44 CLBLM_M_D4 }   { SL1BEG0 IMUX24 CLBLM_M_B5 }  FAN_ALT0 FAN_BOUNCE0 IMUX28 CLBLM_M_C4 }   [get_nets {load_dato_TMR_0_124}]

Route from RapidSmith2
> {CLBLM_R_X39Y97/CLBLM_L_AQ CLBLM_R_X39Y97/CLBLM_LOGIC_OUTS0 {INT_R_X39Y97/SL1BEG0 INT_R_X39Y96/IMUX24  CLBLM_R_X39Y96/CLBLM_M_B5} INT_R_X39Y97/SR1BEG1 {INT_R_X39Y96/IMUX28 CLBLM_R_X39Y96/CLBLM_M_C4} INT_R_X39Y96/IMUX44  CLBLM_R_X39Y96/CLBLM_M_D4}

It looks like the source of this error is that the function  (see code snippet of function below) simply looks for connections which connect to wires contained in the Set<String> wiresInNet. If it finds a connection that hooks into a wire contained in this set, it will be added as the route, whether or not it is the real route.

` 

	for (Connection conn : routeTree.getWire().getWireConnections()) {
				
		Wire sinkWire = conn.getSinkWire();
				
		if (wiresInNet.contains(sinkWire.getFullWireName()) && !visited.contains(sinkWire)) {
			connectionCount++;
			RouteTree sinkTree = routeTree.addConnection(conn);
			searchQueue.add(sinkTree);
			visited.add(sinkWire);
		}
	}
`

In the example above, because INT_R_X39Y96/IMUX28 can also connect to INT_R_X39Y97/SR1BEG1, the importer will hook it into that net, even though the actual route connects to FAN_ALT0 FAN_BOUNCE0.

ttown523, 2017-07-12:
Pull request #249 should fix this. I updated the routing to use PIPs instead of wires. Once I am done defending my thesis I will work on getting this PR merged (with the corresponding Tincr PR). You will need to re-generate the RSCPs when these changes are ready.

trharoldsen, 2017-08-19:
Can we close?

ttown523, 2017-08-19:
I think so, but we need to confirm with @mjc (Matthew Cannon)

ttown523, 2017-08-19:
I meant @mjc31511

mjc31511, 2017-09-15:
From what I can tell the issue has been resolved.

trharoldsen, 2017-09-15:
Appears to be resolved.

closed 2017-09-15 by trharoldsen