byuccl/RapidSmith2 #259: Example Programs Broken by #249

nelsobe, 2017-08-11:
When TCP's were changed to RSCP's in RS2 some time ago (pull request #249), none of the example programs such as PlacerDemo or AStarRouterDemo were updated to reflect that change.  The example programs were thus left in a broken state by #249.  

If we are to provide example programs they should always be functional.  To ensure this they should be part of the test suite so that new users don't waste a lot of time trying to figure out why they don't work on their machines (not knowing that they actually don't work on anybody's machines in their broken state).

ttown523, 2017-08-11:
Hmm interesting. I thought that these worked fine before merging but I will fix them when I get back. I will also add all examples to the test suite so they are all run before merging code. Thanks for catching this!

nelsobe, 2017-08-11:
I didn't actually try to run them so I am only guessing they were broken :-).  I just noticed that they were trying to load RSCP's from .tcp directories which seemed pretty obvious...  I have made changes to them in PR #253 that at least fixes that problem (but I didn't test them either yet).  

I think the best approach will be to add them to the test suite.  Some of them (like maybe the placer demo) might need to modified somewhat to be able to be run stand-alone by the test suite.

trharoldsen, 2017-08-19:
Was this fixed in the last pull request?  Can we close it?

ttown523, 2017-08-19:
This is fixed in #253

closed 2017-08-19 by ttown523