byuccl/RapidSmith2 #260: SitePins of BelPins doesn't factor in RouteThroughs

trharoldsen, 2017-08-19:
BelPin::getSitePins is supposed to provide the site pins that drive the BelPins.  The returned value though does not factor in BEL routethroughs (it is generated directly from the primitive defs info).  Is this the behavior we want or do we want to account for BEL routethroughs?

ttown523, 2017-08-19:
We should probably factor in routethroughs as well. I'm assuming this will be an update to the device generator?

trharoldsen, 2017-08-19:
Yes, this would be in the device genrator.

nelsobe, 2017-08-23:
Issues to keep in mind:
1. This routine is currently not being used anywhere in the code base we can find.
2. While XDLRC files do contain information on site route-throughs (A1 input --> A output pin on a slice), they do not contain information on BEL route-throughs (A1 bel pin --> O6 bel pin).  That information is represented in the familyInfo.xml file.  However, that information is not being queried by this routine and therefore what is returned does not reflect possible BEL route-through paths.

trharoldsen, 2017-12-23:
I vote we remove the subroutine.  Anyone opposed to this solution?

closed 2017-12-27 by trharoldsen