byuccl/RapidSmith2 #272: Vivado Interface does not generate correct route string for long lines

mjc31511, 2017-09-13:
When trying to import a design generate using RapidSmith2 I noticed that nets with routes including long lines would not import. It looks like RapidSmith is using the correct wires, but it does not identify the correct node, or includes wires that are not nodes. Here is a summary of what I have found:
'LV_L9' - Wire not needed (its parent, which is already in the route string, is the node which connects to the sink)
'LV9' - Same as above
'LH6' - Same as above
'LH0' - Wrong node (this is a wire not the node that Vivado expects)
'LV_L0' - Same as 'LH0'

This was observed using Vivado 2016.4/2017.1 (I used both to see if it was a version issue).

trharoldsen, 2017-11-01:
@mjc31511 Can you confirm that pr #294 fixes this issue so we can close this.

nelsobe, 2017-11-03:
Was this fixed in #294?

mjc31511, 2017-11-06:
I'm working on integrating and testing the changes of #294. It'll take about a week to test.

mjc31511, 2017-11-17:
I have tested the changes of #294. It looks like the long line issues were resolved. However, there are now issues with atennas on some of the lines. I have talked to @DallonTG and he thinks he knows what the cause of this issue is.

closed 2017-11-29 by trharoldsen