byuccl/RapidSmith2 #273: Remove ZRGViewer Execution from DotFilePrinterDemo

DallonTG, 2017-09-14:
Currently, the Dot File Printer Demo generates a dot file of a netlist and then attempts to execute ZRGViewer to display the resulting dot file.

Out of the box, ZRGViewer does not actually open on Ubuntu 16.04 and Mac. (Maybe it works on Windows?) On Ubuntu, I copied the exact string that the demo tries to use to start ZRGViewer into the terminal and executed it. When I did this, ZRGViewer actually did open up and eventually show the dot file.

The point of the demo isn't to open up a dot file viewer, so we should remove the automatic ZRGViewer execution from the main function of the demo. If we decide against that, we could maybe fix it and at least make it an optional function like the displayDotfileMac function.

nelsobe, 2017-09-19:
I agree that tracking DOT file viewers on all systems will only lead to broken code in the future.  And, I believe that code tries but fails to reliably open a viewer is worse than code that doesn't even try.

trharoldsen, 2017-11-01:
Any update on this?

nelsobe, 2017-11-03:
This is handled by #299.

closed 2017-11-03 by nelsobe