byuccl/RapidSmith2 #276: Compatible types are null in load device file

trharoldsen, 2017-10-01:
Using the default device file for artix7 included in the repo <xc7a100tcsg324_db>, the compatible_types field in the site templates.  Can someone else confirm this error.

DallonTG, 2017-10-04:
Professor Nelson and I confirmed the error. We're going to look into it some more.

DallonTG, 2017-10-06:
In case you need a fix quickly, the updated device files for the included artix7 device can be found in the "nullCompatibleTypeFix". This fixes the issue, as mentioned above.

I am currently generating an xdlrc for the included kintex device. Once this finishes, I will add the device files for it to the same branch and I'll make a pull request.

DallonTG, 2017-10-13:
Addressed with #284

closed 2017-10-13 by DallonTG