byuccl/RapidSmith2 #280: DesignAnalyzer Multiple Intersite RouteTree Assertion Error

DallonTG, 2017-10-09:
Running the DesignAnalyzer example with the simon.rscp design results in this assertion occurring:
`assert(n.getIntersiteRouteTreeList().size() <= 1);`

Here's the output + trace:
`Net: hscore2_bin_reg[6]_i_3_n_0
  Source Pin: hscore2_bin_reg[6]_i_3.CO[3]
  Pin:  hscore2_bin[6]_i_1.I0
  Pin:  state_reg[2]_i_4.I1
  Pin:  arrow_addr[11]_i_1.I3
  Pin:  state_reg[1]_i_2.I5
Exception in thread "main" java.lang.AssertionError
	at edu.byu.ece.rapidSmith.examples.DesignAnalyzer.prettyPrintDesign(DesignAnalyzer.java:156)
	at edu.byu.ece.rapidSmith.examples.DesignAnalyzer.main(DesignAnalyzer.java:59)`

The program is written with the assumption that regular nets (not VCC or GND nets) will always have a single intersite route tree. However, looking at the simon design in the Vivado gui, this does not always seem to be the case.

This image shows the beginning of the : hscore2_bin_reg[6]_i_3_n_0 net. There is only one source pin, but the net leaves the site at two different site pins, (COUT and DMUX).
[image]

nelsobe, 2017-10-09:
This is correct - the code assumes this will not happen.  But, it was discussed during the writing of this demo program because it was obvious it could happen.  @ttown523 - what are your thoughts?

trharoldsen, 2017-10-09:
The above circuit is completely valid.  There might be a site routethrough
from COUT to DMUX that wasn't used in this case though.