byuccl/RapidSmith2 #285: Cleanup .rscp's and .dcp's after RSCP Import Tests

DallonTG, 2017-10-17:
When gradelw test (or gradelw build) runs, all the .rscp's the ImportTests uses are copied from src/test/resources/test/ImportTests/RSCP/ to build/resources/test/ImportTests/RSCP.

Additionally, all of the DCP files that are used by DesignVerificationTests are copied from src/test/resources/test/ImportTests/DCP to build/resources/test/ImportTests/DCP. 

If you're using Eclipse, these files also get copied at some point under the bin/resources/test/ImportTests directory. 

There's no need for these files to be copied and left after the tests run, so we should delete them afterwards.

trharoldsen, 2017-10-18:
The reason they are copied is so that they can be packaged into a jar to
distribute.  This is the preferred behavior so that we can just jar up the
build directory and everything necessary to run RS is packaged.  We don't
have to then create any hard links to the RS2 source code directory
(outside of the device files directory).

DallonTG, 2017-10-19:
I hadn't thought of that. Thanks a lot for taking the time to explain the reasoning behind this!

closed 2017-10-19 by DallonTG