byuccl/RapidSmith2 #292: PseudoPins on internal cells cannot be disconnected from net.

trharoldsen, 2017-10-26:
Pseudopins can be added to internal cells, but cannot be removed.  This is important to repack a slice.

Example:
```java
CellPin pseudoPin = internalRamLut.attachPseudoPin("A6Pin-Pseudo", PinDirection.IN);
cellNet.attachPin(pseudoPin);
cellNet.disconnectFromPin(pseudoPin); // error here
```

Error message:
```
Exception in thread "main" edu.byu.ece.rapidSmith.util.Exceptions$DesignAssemblyException: Cannot 
remove internal pin from net! remove the external macro pin instead
at edu.byu.ece.rapidSmith.design.subsite.CellNet.disconnectFromPin (CellNet.java:345)
```

The offending line is the CellPin.isInternal check.  isInternal returns true if the cell the pin is on is an internal cell.  The pseudopins, however, are not part of the macro structure and should be treated differently.

trharoldsen, 2017-10-26:
Resolved in #293

closed 2017-10-26 by trharoldsen

nelsobe, 2017-10-26:
Thanks for handling this one!