byuccl/RapidSmith2 #295: RAMD type cells have incomplete placement information.

trharoldsen, 2017-10-27:
The library cells for RAMD32/E and RAMD64/E only list the D6 and D5 (for D32 types only) LUTs as possible placement BELs.  In reality, every LUT in the SLICEMs should be available for these cells.  This is likely a tincr issue, but it affects RS2 so I'm posting it here.

As a note, my guess the reason the TINCR code isn't generating this information correctly is that the RAMs cannot be placed at another cell unless certain conditions are met (eg the D5/6 LUT in the slice must be used by another LUTRAM and the WADDR and RADDR ports must be the same.

trharoldsen, 2017-10-27:
I'll post a quick manual fix for now, but this is something that needs to be looked into.

nelsobe, 2017-10-27:
I have looked through the TIncr code and it determines the list of BELs to include for a cell in the cellLibrary.xml file by doing the following in routine ::tincr::cells::is_placement_legal { cell bel }:

1. Place the cell onto a BEL
2. If no exception thrown by Vivado _AND_ the BEL it was actually placed onto is the same specified in step 1, then return that it is a legal placement. 

For LUTRAMs, I believe that when they are placed onto anything other than the D Luts, Vivado moves them to the D Lut and thus the routine returns false.

@ttown523 - do you remember if the movement only happens with LUTRAMs or does it happen with other cells as well?  That may affect how/where we put in a fix for it.

The obvious options I see would include:

- Treat LUTRAMs as a special case in this routine.
- Put custom code elsewhere into the cellLibrary.xml generation code to add what @trharoldsen  has added above
- Use Travis' temporary fix long-term

trharoldsen, 2017-11-01:
Patched the file in the repo as a short term fix.  I don't think the change is a good long term fix however as it would need to be done for each device file.