byuccl/RapidSmith2 #302: Incorrect SiteTypes for xc7a200tsbg484 device

mjc31511, 2017-11-06:
It looks like there are some inconsistencies within how RapidSmith2 represents a Site's SiteType. To give some context, I am moving sites from SLICEM types to SLICEL types (only if it can be validly placed). I get a NullPointerException in the following line (in Site.java)

return externalWires.get(type).get(pinName);

in the getExternalWire method. The call externalWires.get(type) returns null for the passed in type. I have found a workaround for this method with the following code:

private int getExternalWire(SiteType type, String pinName) {
		if(externalWires.get(type) == null){
			if(externalWires.get(SiteType.valueOf(type.family(), "SLICEL")) == null){
				return externalWires.get(SiteType.valueOf(type.family(), "SLICEM")).get(pinName);
			}
			else{
				return externalWires.get(SiteType.valueOf(type.family(), "SLICEL")).get(pinName);
			}
		}
		return externalWires.get(type).get(pinName);
	}
However, this is just a workaround. The call stack (provided below) should provide the correct type to the slice. 

Exception in thread "main" java.lang.NullPointerException
        at edu.byu.ece.rapidSmith.device.Site.getExternalWire(Site.java:658)
        at edu.byu.ece.rapidSmith.device.Site.getSitePin(Site.java:608)
        at edu.byu.ece.rapidSmith.device.Site.getSitePin(Site.java:589)
        at edu.byu.ece.rapidSmith.design.subsite.CellPin.mapToBelPin(CellPin.java:198)
        at edu.byu.ece.crossDomain.placer.AutoPlace.placeDesign(AutoPlace.java:424)

As an example, the site SLICE_X40Y136 says it is a SLICEL (and SLICEL is the type passed in), but in Vivado it is a SLICEM. There are many more examples of this.

trharoldsen, 2017-11-06:
The type of the site is a mutable property and can change to reflect how the site is being used.  When placing "instances" onto the sites, the type of the site should be updated with the correct site type.  This will update all of the site type specific properties for the site.  

The type can be set with the setType() method on the Site object.  Note, I don't think we check whether the proposed site type is a legal type for that site.  When loading the designs, the types of the sites are automatically updated to the type of the "instance" located on the site.  I presume this is why you are seeing a type of SLICEL on a SLICEM site.  Simply placing a cell onto the site, though, will not change the site's type.

To find the possible types that a site can be configured as, use the Site::getPossibleTypes() method.

mjc31511, 2017-11-07:
Good catch. A problem with using old code is that I didn't know it was setting the SiteType to be an illegal value. However, I would argue for a check to make sure you are not setting the Site to an illegal type. Maybe in the form of an assertion would be appropriate.

trharoldsen, 2017-12-27:
The default behavior in the latest master branch checks for invalid types.  If the unsafe update method is desired for speed reasons, the user can use the setTypeUncheck method.

closed 2017-12-27 by trharoldsen