byuccl/RapidSmith2 #304: Design::deepCopy does not work correctly

trharoldsen, 2017-11-08:
Method was not updated for new structure like pseudopins and macros.

nelsobe, 2017-12-13:
Note: be sure to address property copying when it is fixed since that does not work in the fixes of #314.

trharoldsen, 2017-12-23:
As @nelsobe mentioned, the deep copy methods may prove troublesome to keep up to date with merges.  Though I don't necessarily agree, there seems to be a willingness to add additional parameters to the base cell classes that will be difficult to keep up the deep copy up to date with.

One solution would be to move deep copy out of the primary clases and into its own class turning it into more of a second class object.  Or we could completely move it over to the rscad repo and only keep what I need.

trharoldsen, 2018-02-10:
What is the status of deep copy?  Do we want to maintain it or remove it?