byuccl/RapidSmith2 #306: Import Error: CellPin connected to net not in design

trharoldsen, 2017-11-12:
On importing the Simon test design, I have found a pin connected to a net that is not in the design.

pin - MemDB_IOBUF[11]_inst/I
net - <const0>

The cell appears to be a macro cell, the internal cells are connected to the global GND net.  I suspect this is a lingering artifact from importing.

trharoldsen, 2017-11-13:
Looking closer at it, since it's giving me problems -- the importer initially connects the macro cells to the original GND/VCC net.  This in turn connects the pins on the internal cells to the net.  while the macro cell pin points to being connected to the net (for reference to the user), the net does not contain the macro cell pin as a pin on the net (only the internal pins are added).

Once the importer is done reading the EDIF, it then consolidates all of the GND/VCC nets into a single net each.  In this process, the GND/VCC nets are disconnected from the design (detached from all pins and removed from the design), and the pins on the net are added to the respective consolidated GND/VCC net.  However, the macro cell pin is not listed as a pin on the net (it was never added, it just points to the net), and therefore, this pin is never updated.  This leaves the macro cell pin connected to a zombie net.

Realistically, I don't think connecting nets directly to the internal cell pins is something that should ever be done in legal operations.  We should probably disable that ability (throwing an error).  All connecting should happen through the macro ports.  I don't know at this time if we have a method of getting from the internal cell pin back to the macro cell pin.  This ability would be a) very useful when traversing/modifying the design, and b) probably necessary to fix this bug as instead of reconnecting each individual internal cell pin, we should be reconnecting the macro cell pin.

trharoldsen, 2017-11-15:
Internal VCC/GND macro nets (those completely encompassed in the macro) and cells have major issues.  The importer wants to replace the internal net with the global VCC/GND and remove the source cell.  However, doing so completely corrupts the macro and puts the design in an awkward (and probably illegal) state.  Trying to remove the preserve the internal cells and macros and just having multiple VCC/GND nets doesn't work either though as the routing for these nets is stored within the global VCC/GND net.

trharoldsen, 2017-11-15:
I'm uploading my current fix to https://github.com/trharoldsen/RapidSmith2-public/tree/macroNetIssue.

Here's the error I'm getting when I run my design copy tests (where this issue is being exposed).

java.lang.AssertionError: Valid intrasite route not found from site pin : BUFGCE_X1Y0/CE_PRE_OPTINV Net: RapidSmithGlobalVCCNet
 
    at edu.byu.ece.rapidSmith.interfaces.vivado.XdcRoutingInterface.createIntrasiteRoute(XdcRoutingInterface.java:716)
    at edu.byu.ece.rapidSmith.interfaces.vivado.XdcRoutingInterface.processIntersitePins(XdcRoutingInterface.java:225)
    at edu.byu.ece.rapidSmith.interfaces.vivado.XdcRoutingInterface.parseRoutingXDC(XdcRoutingInterface.java:146)
    at edu.byu.ece.rapidSmith.interfaces.vivado.VivadoInterface.loadRSCP(VivadoInterface.java:108)
    at edu.byu.ece.rapidSmith.interfaces.vivado.VivadoInterface.loadRSCP(VivadoInterface.java:52)
    at design.rscpImport.ImportTests.flappyUltrascaleTest(ImportTests.java:112)
    at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
    at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
    at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
    at java.lang.reflect.Method.invoke(Method.java:497)
    at org.junit.platform.commons.util.ReflectionUtils.invokeMethod(ReflectionUtils.java:389)
    at org.junit.jupiter.engine.execution.ExecutableInvoker.invoke(ExecutableInvoker.java:115)
    at org.junit.jupiter.engine.descriptor.TestMethodTestDescriptor.lambda$invokeTestMethod$6(TestMethodTestDescriptor.java:167)
    at org.junit.jupiter.engine.descriptor.TestMethodTestDescriptor$$Lambda$194/931675031.execute(Unknown Source)
    at org.junit.jupiter.engine.execution.ThrowableCollector.execute(ThrowableCollector.java:40)
    at org.junit.jupiter.engine.descriptor.TestMethodTestDescriptor.invokeTestMethod(TestMethodTestDescriptor.java:163)
    at org.junit.jupiter.engine.descriptor.TestMethodTestDescriptor.execute(TestMethodTestDescriptor.java:110)
    at org.junit.jupiter.engine.descriptor.TestMethodTestDescriptor.execute(TestMethodTestDescriptor.java:57)
    at org.junit.platform.engine.support.hierarchical.HierarchicalTestExecutor.lambda$execute$3(HierarchicalTestExecutor.java:83)
    at org.junit.platform.engine.support.hierarchical.HierarchicalTestExecutor$$Lambda$155/817978763.execute(Unknown Source)
    at org.junit.platform.engine.support.hierarchical.SingleTestExecutor.executeSafely(SingleTestExecutor.java:66)
    at org.junit.platform.engine.support.hierarchical.HierarchicalTestExecutor.execute(HierarchicalTestExecutor.java:77)
    at org.junit.platform.engine.support.hierarchical.HierarchicalTestExecutor.lambda$null$2(HierarchicalTestExecutor.java:92)
    at org.junit.platform.engine.support.hierarchical.HierarchicalTestExecutor$$Lambda$158/922511709.accept(Unknown Source)

nelsobe, 2017-11-29:
Phone discussion with Travis 11/28:  (Travis please correct/comment)

- The key is to only have 1 VCC and 1 GND in a CellDesign.  
- Circuits come in from Vivado with many VCC's and many GND's but Thomas' post processing of the design on import collapses all the VCC nets into one and all the GND nets into one.
- However, in the case of adding new cells or adding macros to the design, additional VCC and additional GND cells get created.
- PROPOSAL: modify the code so that each design has a dedicated VCC and a dedicated GND cell.
- When cells are added to the design, do not add new VCC or GND cells, just return the handle to the designs's existing VCC or GND cell.  
- We cannot prevent users from creating new VCC or GND cells, but when they are attempted to be added to the design they can be ignored and a handle returned to the design's dedicated VCC or GND.  
- Or, an error could be thrown to notify the user.  I like this second approach better lest the user retain a handle to a VCC or GND cell which is not being used and then try to route signals to it or otherwise use it.
- Another idea is to have the Cell constructor throw an error if a user tries to create a VCC or GND cell and provide an explanatory message.  I like this equally well.  However, we would need a way for the import code to create a VCC cell and a GND cell.
- Regardless, this will require adding two new members to the CellDesign class - a VCC cell and a GND cell and a bunch of code modifications to ensure they are the only VCC and GND ever in the design.

trharoldsen, 2017-11-29:
I should add that I believe the preferred method would be to allow multiple VCC/GND cells in a design and correctly route the nets coming from each.  However, with what we can get out of Vivado, this may not be possible.

trharoldsen, 2017-12-23:
Resolved in #317.  The current fix is probably not the best approach, but it is the quickest and safest fix for now.

closed 2017-12-23 by trharoldsen