byuccl/RapidSmith2 #307: Support for Zynq UltraScale+ family

sch-alex, 2017-11-14:
Are there any plans to add support for the Zynq UltraScale+ family in the near future?

kru-alex, 2017-11-15:
Good point. That would also be very important for one of our projects.

DallonTG, 2017-11-22:
Sorry for the delay in responding. We are planning on eventually adding support for the Zynq Ultrascale+ family, but this will unfortunately probably not happen in the near future.  

Section [10.2 of the Tech Report](https://github.com/byuccl/RapidSmith2/blob/master/docs/TechReport/TechReport.pdf) explains how you can add work to add support for a new family. You can follow these instructions if you wish and we will happily provide support to you along the way.

Otherwise, we will comment on this issue when we are eventually actively working on adding support for the Zynq Ultrascale+ family.