byuccl/RapidSmith2 #354: RS2: BSCAN primitive definition not found! Please add this file and rerun.

amitkulkarnis, 2018-08-17:
Hi,
When I try to build a new device file for Zynq (device names: xc7z020clg484-1 and xc7z045ffg900-2) I get the following error: 
Percent complete: 100% 
ERROR: BSCAN primitive definition not found! Please add this file and rerun.

Where can I get this primitive definition?

Thanks.

DallonTG, 2018-08-17:
This is a [tincr ](https://github.com/byuccl/tincr/) problem. In the past, we thought that the primitive definitions, which describe site types, were consistent within xilinx families. Because of this, Tincr has folders for primitive definitions for families under the cache directory. However, we found out that this does not seem to be the case. You can see some discussion on this [here](https://github.com/byuccl/tincr/pull/72). We haven't implemented a fix for when devices within a family have different primitive definitions yet.

Without trying to generate the XDLRCs for the two zynq devices you mentioned, it looks like they probably don't share the same exact primitive definitions. 

To help you, I've added a Tincr branch that should contain the correct primitive definitions for xc7z020clg484-1. Can you checkout the[ zynqGroup1 tincr branch](https://github.com/byuccl/tincr/tree/zynqGroup1) and try to create an XDLRC for the xc7z020clg484-1 part? This should work, so please let us know if it doesn't.

Then to create the XDLRC for the xc7z045ffg900-2 part, please checkout the [zynqPdefs branch(https://github.com/byuccl/tincr/tree/zynqPdefs) first.

I realize this isn't a permanent solution and that the problem will come up for other parts. But please try this out and let me know if it works. Thanks.

JerryMain, 2018-10-03:
I had the same error for the xc7z020clg484-1, after switching to the [zynqGroup1 tincr branch](https://github.com/byuccl/tincr/tree/zynqGroup1) no error was reported.

However after trying to generate the cellLibary for the xc7z020clg484-1 the following error was reported:
`ERROR: [Vivado 12-4885] Site IDELAY_X1Y147 cannot be sitetype IDELAYE2_FINEDELAY. It must be one of: IDELAYE2`
I don't know if this is a related issue or another issue altogether, however if you have any ideas about that please let me know.

Thanks.

trharoldsen, 2018-10-03:
With only a perfunctory glance, my guess is that the IDELAYE2_FINEDELAY 
primitive is not present as an alternate site type of the IDELAYE2 
primitive in the familyInfo.xml which was used to build the device file 
for the part.

DallonTG, 2018-10-03:
I apologize that these issues haven't been documented well for everyone to know of. I exchanged a few emails with Amit about these issues.

I don't think generating the cell library depends on the family info xml. Amit reported seeing the same error when trying to generate the cell library. He was using Vivado 2018.2. I don't see this error and I am using Vivado 2016.2. What version of Vivado are you using? This issue may only occur in newer versions.

In any case, I would actually recommend you don't try to generate a new cell library for xc7z020clg484-1. I know the documentation currently suggests making a new one for each part, but our working assumption has been that the cell library should be the same for the entire zynq family. Additionally, Zynq and 7 Series cell libraries require some manual edits. The zynq cell library already included with RapidSmith2 should already include these manual edits.

JerryMain, 2018-10-04:
> I don't think generating the cell library depends on the family info xml. Amit reported seeing the same error when trying to generate the cell library. He was using Vivado 2018.2. I don't see this [...]

I'm using Vivado 2017.2, so that is indeed a newer version.

> In any case, I would actually recommend you don't try to generate a new cell library for xc7z020clg484-1. I know the documentation currently suggests making a new one for each part, but our working assumption has been that the cell library for all devices in a family should be the same. The zynq cell library already included with RapidSmith2 should already include these manual edits. So, I would just skip the cell library generation step for xc7z020clg484-1.

For now I'll work with the cell library that is included.
Thank you for the reply and suggestion.

closed 2019-01-15 by DallonTG