byuccl/RapidSmith2 #355: No bels matched *RAMB36E1 when creating pinmapping.

trharoldsen, 2018-09-18:
With xc7a200, I was trying to create the pin mappings for a RAMB36E1 cell placed on a RAMB36E1 BEL.  When querying for a RAMB36E1 BEL, Vivado/Tincr only lists the RAMBFIFO36E1 BELs (I believe this is because the RAMB36E1 BELs only show up after a BEL is placed?).  As such, the cell does not get placed and the script fails to make the pin mappings.

nelsobe, 2018-09-18:
Can you provide some clarification on this?  Are you saying that the netlist has a RAMB36E1 cell but when you query you cannot find any RAMB36E1 BEL’s to place it on (and that all you can find are RAMBFIFO36E1 BELs)?  Or is it that you find the BEL but no pin mappings exist?

This might have to do with alternate site types.  Or, it might have to do with dynamic pin mappings.  Or, maybe both :-(

If you can help clarify things a bit I will do some digging tomorrow and get back with you (I have worked on both topics a lot in the past so should be able to help).

trharoldsen, 2018-09-18:
It's a combination of both.  This issue cropped up with dynamic pin mappings.  When trying to create the correct pin mapping, the PinMapping creation code was failing doing to BELs in the the alternate site type not being listed.

trharoldsen, 2018-09-18:
The patch is up under #356

trharoldsen, 2018-09-20:
Fixed is merged.  Closing

closed 2018-09-20 by trharoldsen