byuccl/RapidSmith2 #364: Design not implementable when only imported and exported but not changed

mthoonen, 2019-01-12:
I am trying to modify an AES algorithm in RapidSmith2, but even when I'm not changing anything the design is not implementable in Vivado 2017.2 anymore. It throws the following error during route_design:

[Place 30-642] Placement Validity Check : Instance  aesDec/t_STATE_RAM0_reg_0_3_0_5/RAMA of type RAMD32 is not Placeable.
Reason:  for bel A6LUT The slice contains LUT-RAMs but none of them occupy the DLUT. A DLUT RAM is needed to provide write address signals to the rest of the RAM,  for bel B6LUT The slice contains LUT-RAMs but none of them occupy the DLUT. A DLUT RAM is needed to provide write address signals to the rest of the RAM,  for bel C6LUT The slice contains LUT-RAMs but none of them occupy the DLUT. A DLUT RAM is needed to provide write address signals to the rest of the RAM,  for bel D6LUT Conflicting nets for SLICE_X30Y149.D6LUT.A2
Two net names are: aesDec/i_RAM_ADDR_RD0[1] and aesDec/i_RAM_ADDR_WR0[1],  for bel A5LUT The slice contains LUT-RAMs but none of them occupy the DLUT. A DLUT RAM is needed to provide write address signals to the rest of the RAM,  for bel B5LUT The slice contains LUT-RAMs but none of them occupy the DLUT. A DLUT RAM is needed to provide write address signals to the rest of the RAM,  for bel C5LUT The slice contains LUT-RAMs but none of them occupy the DLUT. A DLUT RAM is needed to provide write address signals to the rest of the RAM,  for bel D5LUT Conflicting nets for SLICE_X30Y149.D5LUT.A2
Two net names are: aesDec/i_RAM_ADDR_RD0[1] and aesDec/i_RAM_ADDR_WR0[1]

and 23 similar errors. The program I run in RapidSmith is the following:
package maartent.netlistcreator;

import java.io.IOException;

import edu.byu.ece.rapidSmith.design.subsite.CellDesign;
import edu.byu.ece.rapidSmith.design.subsite.CellLibrary;
import edu.byu.ece.rapidSmith.device.Device;
import edu.byu.ece.rapidSmith.interfaces.vivado.VivadoCheckpoint;
import edu.byu.ece.rapidSmith.interfaces.vivado.VivadoInterface;

public class QuickImportExport {

	private static CellDesign design;
	private static Device device;
	private static CellLibrary libCells;

	public static void main(String[] args) throws IOException {
		VivadoCheckpoint vcp = VivadoInterface.loadRSCP(args[0]);
		QuickImportExport.design = vcp.getDesign();
		QuickImportExport.device = vcp.getDevice();
		QuickImportExport.libCells = vcp.getLibCells();	
		String outFileName = args[0].substring(0, args[0].length() - 4) + "tcp";
		VivadoInterface.writeTCP(outFileName, design, device, libCells);
		System.out.println("Done!");
	}

}

(essentially a slightly simplified ImportExportExample)

The device I'm targeting is the zedboard, with FPGA part number XC7Z020-CLG-484-1. The design is implementable when building directly from VHDL and XDC files in Vivado but not after writing an RSCP and running the program above. One thing I noticed is that the amount of LUTs grows with about 80 LUTs more after exporting and importing. Any idea what might cause the error? If linking the VHDL and XDC helps I could do that, what would be the best way to do that?

mthoonen, 2019-01-13:
Took me a while to get 2016 installed, apparently some issues with win 10...
I now get an error while importing:
`Exception in thread "main" edu.byu.ece.rapidSmith.util.Exceptions$ParseException: Site "IOB_X1Y62" not found in the current device. 
On line 3423 of c:\users\maarten\desktop\vivado\AES2016.rscp\placement.rsc
	at edu.byu.ece.rapidSmith.interfaces.vivado.XdcPlacementInterface.tryGetSite(XdcPlacementInterface.java:261)
	at edu.byu.ece.rapidSmith.interfaces.vivado.XdcPlacementInterface.applyCellPlacement(XdcPlacementInterface.java:112)
	at edu.byu.ece.rapidSmith.interfaces.vivado.XdcPlacementInterface.parsePlacementXDC(XdcPlacementInterface.java:93)
	at edu.byu.ece.rapidSmith.interfaces.vivado.VivadoInterface.loadRSCP(VivadoInterface.java:106)
	at edu.byu.ece.rapidSmith.interfaces.vivado.VivadoInterface.loadRSCP(VivadoInterface.java:52)
	at maartent.netlistcreator.QuickImportExport.main(QuickImportExport.java:18)
`

The design is routed, so I guess that was the cause for increased LUTs. The placement.rsc file is linked below, I can't generate a TCP as RapidSmith now gives an error.
https://drive.google.com/file/d/1w7ptuMfd5VqWfvBFUDXCtkDjz01dZpjs/view?usp=sharing

trharoldsen, 2019-01-13:
What device is this design for?

mthoonen, 2019-01-13:
XC7Z020CLG484-1

mthoonen, 2019-01-15:
update: I also tried in on the XC7a100 device from the tech report, and it gives the same problem. It seems that vivado creates some variation of LUT ram during synthesis that some part of the chain can't handle.

DallonTG, 2019-01-15:
There may have been some TCP reordering, but I don't know of any changes that were made for LUT RAMs.

It seems like there are two issues here: a LUT RAM issue and an issue with the zynq device. You only encountered an issue with the XC7a100 device when re-importing into Vivado, correct? If you could run read_tcp with the -verbose option, that would be helpful. I'm expecting this will let us see some warnings that occur as the placement.xdc is parsed.

DallonTG, 2019-01-15:
I have tried to replicate the issue with some designs that use LUT RAMs on the xc7a100 device, but have been unable to. @mthoonen, can you generate a TCP with the xc7a100 device? Or is that only an issue with the zynq device?

nelsobe, 2019-01-16:
I have been following the message trail here and I think it would be useful if you could summarize just what the current state is so we could understand what might be the issue in order to help.  

From what you provide above, it is not clear if there are still errors.  Do you have a design that is still not working correctly?  If so, can you summarize the details on versions of the tools you are using and give the error messages you are receiving?  And, if they are not private, a copy of the TCP would be useful as well.

Thanks!

mthoonen, 2019-01-16:
I have an AES design that implements in Vivado (2016.2 and 2017.2), but if I export it as an RSCP, import it into Rapidsmith, change nothing and then export as TCP it gives the error from the first message when synthesizing the design in Vivado:

> [Place 30-642] Placement Validity Check : Instance aesDec/t_STATE_RAM0_reg_0_3_0_5/RAMA of type RAMD32 is not Placeable.
> Reason: for bel A6LUT The slice contains LUT-RAMs but none of them occupy the [...]

I'm using RapidSmith 2 2.0.0. Do you need the TCP generated by rapidsmith or directly from Vivado? 
Here is the tcp generated by rapidsmith: https://drive.google.com/drive/folders/1_os7vVVxRQdFrDoOzs2xpXyuccTgNRQs?usp=sharing
And here by Vivado: https://drive.google.com/drive/folders/1YDmdfxmLNzROfMHHx-G1IWCYyTdOMzy-?usp=sharing

nelsobe, 2019-01-16:
Thanks.  One question - the documented flow is:

Vivado --> circuit.rscp --> RapidSmith2 (do some processing here) --> circuit.tcp --> Vivado

How did you get the .tcp that you say was generated by Vivado?  The write_tcp command is deprecated and has been replaced by write_rscp.

What would be useful is to get the .rscp generated by Vivado and also the .tcp generated by RapidSmith2.

Thanks.

mthoonen, 2019-01-16:
Thanks for the quick reply and your time! I created a tcp both from rapidsmith and with write_tcp. 
Anyway, here is a folder with the RSCP created by write_rscp in Vivado 2016.2 and the TCP created by RapidSmith2 2.0.0: https://drive.google.com/drive/folders/1MrEEr4tlxqghuto3xlIexEq3pQSr-pIx?usp=sharing

DallonTG, 2019-01-16:
Thanks for providing those files for us.

I am now able to reproduce this error when I export a synthesized (but not placed) design with RAM32 LUT RAMs from Vivado as an RSCP. After importing it into RS2, exporting it as a TCP, and then re-importing back into Vivado, I get the same errors. Before the errors there is this warning message:
`WARNING: [Place 30-879] Found overlapping PBlocks. The use of overlapping PBlocks is not recommended as it may lead to legalization errors or unplaced instances.`

I will keep looking to see if I can find a fix.

DallonTG, 2019-02-06:
@mthoonen This should be resolved with #365. If you find that there is still a problem, please re-open this issue!

Thanks for reporting this problem.

closed 2019-02-06 by DallonTG