byuccl/RapidSmith2 #387: Unable to verify my Rapidsmith installation

sandeeprpandey, 2020-05-08:
Dear all,
I tried to install Rapidsmith2 using the "TechReport.pdf"
[TechReport.pdf](https://github.com/byuccl/RapidSmith2/files/4599894/TechReport.pdf)
I could finish 3.1.2 i.e steps for installation given in the  "TechReport.pdf". I am using Eclipse (Version: 2020-03 (4.15.0)          Build id: 20200313-1211).

[image]

Following is the configuration in Eclipse(run configurations):
[image]

[image]

[image]

[image]

When i click the run button, i get following error:

[image]

I am doing this all for the first time. Please excuse me and help if possible. I am using windows 10.

trharoldsen, 2020-05-08:
I think you are using the 64 bit version of Java.  Unfortunately, on
windows, only the 32 bit version of qtjambi library was ever built so
you'll need to use a 32 bit version of Java.

sandeeprpandey, 2020-05-09:
thank you trharoldsen for the reply...
now i downloaded "jdk-8u251-windows-i586". I have set following as environmental variables
[image]

I removed older 64bit java. When i try to redo steps for installation from 3.1.2, i get following 
[image]

Any help??

sandeeprpandey, 2020-05-09:
i restarted the computer.....now i am getting this error:

[image]

Any help...please...

trharoldsen, 2020-05-09:
I suspect you need to update your JAVA_HOME environment variable to your
new install (probably in C: Program Files (32)/...

sandeeprpandey, 2020-05-11:
> 
> I suspect you need to update your JAVA_HOME environment variable to your new install (probably in C: Program Files (32)/...
> […](#)

My JAVA_HOME variable is pointing to correct path..I think...
[image]

I even tried on Ubuntu....But when i run **./gradlew build**, i get following error..
[image]

trharoldsen, 2020-05-11:
Looks like a heap space issue.

Try adding the following line to your gradle.properties file.

org.gradle.jvmargs=-Xmx2000m

trharoldsen, 2020-05-11:
As for Ubuntu, everything looks like it built.  The error seems to be in
the unit testing.  I'm not sure what tests failed (they may be related to
interfacing rapidsmith with vivado though).  Either way, rapidsmith should
be completely usable on your ubuntu install.

sandeeprpandey, 2020-05-12:
> 
> Looks like a heap space issue. Try adding the following line to your gradle.properties file. org.gradle.jvmargs=-Xmx2000m On Sun, May 10, 2020 at 11:08 PM sandeeprpandey <notifications@github.com> wrote:
> […](#)

This is what my "gradle.properties" looks like:

[image]

and after building it, still getting the same problem:
[image]

DallonTG, 2020-05-12:
Pretty sure the ReverseWireTests are failing, though they did not fail in the original post. I have noticed that these tests take quite a while in the past. 

Also, I think you should only need the 32 bit version of Java if you're trying to run a GUI executable. Otherwise, you can (and probably should) use a 64-bit version. I think you might just want to go back to your original setup.

What executable were you trying to run originally?

sandeeprpandey, 2020-05-12:
Thanks DallonTG for the reply...
You mean "originally" i.e. windows10 with 64 bit java.....??